sbwml/r4s_build_script #2: 使用 Github Actions 构建的问题

Ladavian, 2023-05-08:
放在github上面编译，最终固件编译完成了，但是不能上传到release，提示Error: Error 422: Validation Failed: {"resource":"Release","code":"custom","field":"tag_name","message":"tag_name is not a valid tag"}, {"resource":"Release","code":"custom","message":"Published releases must have a valid tag"}, {"resource":"Release","code":"invalid","field":"target_commitish"}

sbwml, 2023-05-08:
github 几率性问题。什么都不用动重新运行任务它可能会成功发布

或者参考 https://github.com/sbwml/builder/blob/master/.github/workflows/build-snapshots.yml#L106 把固件文件打包到 [actions 页面下载](https://github.com/sbwml/builder/actions/runs/4908124814)  ，ps：参考≠复制粘贴

sbwml, 2023-05-09:
你这个任务流改的稀烂

reset 回到  2de7bdcedec0a211a71829cbe2f110b74f207791 这个提交，修改 104 行的  `master` 为你分支名称 `main`

https://github.com/Ladavian/r5c/blob/2de7bdcedec0a211a71829cbe2f110b74f207791/.github/workflows/build-release.yml#L104

closed 2023-05-10 by sbwml