sbwml/r4s_build_script #3: 麻烦帮忙看看~

Pacio, 2023-08-14:
[image]
编译完了上传提示目录没文件~下面还有一个token访问权限问题

Pacio, 2023-08-14:
> [ec6d95e](https://github.com/sbwml/r4s_build_script/commit/ec6d95ea12f5236a8bd500fc3df4040e5d276477)
> 
> 按这样改改就好了
> 
> 至于release 权限问题，你得在仓库添加 Actions 令牌 https://github.com/sbwml/r4s_build_script#%E4%BD%BF%E7%94%A8-github-actions-%E6%9E%84%E5%BB%BA

token添加了啊~不知道为什么还报错
[image]

sbwml, 2023-08-14:
可能你创建的token 并没有给到 actions 足够的权限

Pacio, 2023-08-14:
> 另外你要自己编译的话，要固件支持 kmod 安装或者使用 daed 科学工具，还需要加上这个参数 [b691310](https://github.com/sbwml/r4s_build_script/commit/b6913104379ac62e850a17e8e6516276a7563473)

ok ,我修改一下，麻烦把“export BUILD_FAST=y BUILD_SDK=y”这个参数也加进去吧，我不太清楚这个应该加在什么位置

sbwml, 2023-08-14:
https://github.com/sbwml/r4s_build_script/commit/b6913104379ac62e850a17e8e6516276a7563473  这个就是加进去了

你可以重新复制一下readme 的模板过去改下用户名就行

Pacio, 2023-08-15:
> [b691310](https://github.com/sbwml/r4s_build_script/commit/b6913104379ac62e850a17e8e6516276a7563473) 这个就是加进去了
> 
> 你可以重新复制一下readme 的模板过去改下用户名就行

嗯，目前模板工作正常了~谢谢，就是有个问题不知道为什么我修改了config-musl-x86以后nlbwmon就不能正常统计流量了,我config里面查了半天没发现有什么相关的依赖被我删除啊
https://github.com/Pacio/r4s_build_script/blob/master/openwrt/23-config-musl-x86
[image]

sbwml, 2023-08-15:
首先这个东西在我这里它工作，其次任何插件你问我都没用，这些不是我能控制的东西。

Pacio, 2023-08-15:
> 首先这个东西在我这里它工作，其次任何插件你问我都没用，这些不是我能控制的东西。

好的，我自己研究下~之前我一直用你的Snapshots分支没遇到这问题，现在自己编译默认的是Releases不知道是不是源码分支切换后产生了兼容问题

closed 2023-08-15 by Pacio