sbwml/r4s_build_script #9: Openwrt x86_64 MHI PCI failed.

mdsdtech, 2024-01-08:
Hi bard!. i have a few question to asked.

Q1. I installed the the below KMOD in the openwrt(x86_64_Releases_Kernel_6.6.9) for 5G modem RM520NGLAP with PCIe. But i got the following error. below i attached the logs. is there any solutions to fix this issues?

KMOD: 
1. kmod-mhi-bus_6.6.9-1_x86_64
2. kmod-mhi-net_6.6.9-1_x86_64
3. kmod-mhi-pci-generic_6.6.9-1_x86_64
4. kmod-mhi-wwan-ctrl_6.6.9-1_x86_64
5. kmod-mhi-wwan-mbim_6.6.9-1_x86_64
6. kmod-qrtr-mhi_6.6.9-1_x86_64

demsg log:
<details>
[    9.820931] mhi-pci-generic 0000:03:00.0: MHI PCI device found: qcom-sdx65m
[    9.821228] mhi-pci-generic 0000:03:00.0: BAR 0: assigned [mem 0xe0000000-0xe0000fff 64bit]
[    9.821588] mhi-pci-generic 0000:03:00.0: enabling device (0100 -> 0102)
[    9.822670] mhi mhi0: Requested to power ON
[    9.822845] mhi mhi0: Power on setup success
[   25.887867] mhi mhi0: Device failed to enter MHI Ready
[   25.888130] mhi mhi0: MHI did not enter READY state
[   25.888443] mhi-pci-generic 0000:03:00.0: failed to power up MHI controller
[   25.889145] mhi-pci-generic: probe of 0000:03:00.0 failed with error -110
</details>

lspci:
<details>
lspci -nn -d ::ff00 -v<br>
03:00.0 Unassigned class [ff00]: Qualcomm Device [17cb:0308]
	Subsystem: Qualcomm Device [17cb:5201]
	Flags: fast devsel
	Memory at e0000000 (64-bit, non-prefetchable) [size=4K]
	Memory at e0001000 (64-bit, non-prefetchable) [size=4K]
	Capabilities: [40] Power Management version 3
	Capabilities: [50] MSI: Enable- Count=8/32 Maskable+ 64bit+
	Capabilities: [70] Express Endpoint, MSI 00
lspci: Unable to load libkmod resources: error -2
</details>

Q2. can you please include the this driver in the feature build?
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/net/wwan/t7xx?h=v6.7

Anyway thank you 🙏 for compiling the awesome firmware.

sbwml, 2024-01-09:
1、我没有办法帮你完成 `5G modem RM520NGLAP` 设备的驱动以及使用，因为我没有这个模块和它的任何资料。因为对于一个未知的东西进行“调试” 效率是极低的，所以没办法帮到你。

2、`drivers/net/wwan/t7xx` 我可以以 `kmod-mtk_t7xx` 内核模块支持的方式提供。但仅限于我当前使用的内核版本（即6.6.x），而非你上述所说的 `tree/drivers/net/wwan/t7xx?h=v6.7` kernel-v6.7

mdsdtech, 2024-01-10:
Hi bard!. i add this comment just feedback for kmod-mtk_t7xx you provided. Everything Works perfectly 👍. The driver working well with the Module MediaTek FM350GL. Thank you m8, i will close this issues.

[image]

closed 2024-01-10 by mdsdtech