sbwml/r4s_build_script #14: nginx-util failed to build

ilxp, 2024-01-23:
日志如下：
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:134: /builder/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/.built] Error 1
make[2]: *** [package/Makefile:128: package/feeds/packages/nginx-util/compile] Error 1
make[1]: *** [package/Makefile:122: /builder/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[3]: Leaving directory '/builder/openwrt/feeds/packages/net/nginx-util'
time: package/feeds/packages/nginx-util/compile#1.82#0.21#2.14
    ERROR: package/feeds/packages/nginx-util failed to build.
make[2]: Leaving directory '/builder/openwrt'
make[1]: Leaving directory '/builder/openwrt'
make: *** [/builder/openwrt/include/toplevel.mk:232: world] Error 2

sbwml, 2024-01-23:
要么同步最新commit   要么就回到这个commit https://github.com/sbwml/r4s_build_script/commit/ca017e377a5c66d616edc522d2bf9c8e1251efe6

ilxp, 2024-01-23:
是同步到了最新的commit

sbwml, 2024-01-23:
我建议直接 fork，这玩意我action 每夜构建，真有问题我会发现的比你早

发自我的iPhone

ilxp, 2024-01-23:
ok，我找到原因了。

closed 2024-01-23 by ilxp