sbwml/r4s_build_script #23: 问一个小问题

minkoonim, 2024-04-03:
要是fork本repo,
在openwrt/files/etc/config,
放自己的配置, 比如network
这样能修改默认配置嘛?

sbwml, 2024-04-04:
不能，这个仓库你看到的所有文件它都是来自一个 caddy2 服务器上的备份。所有的目录结构仅用于方便区分文件用途，而不是直接针对 openwrt 工程目录。

你能上传你自己的配置到 r4s_build_script/openwrt 位置，在任意一个 sh 文件中通过 curl 或 wget 下载到 openwrt 工程位置上，如 

https://github.com/sbwml/r4s_build_script/blob/master/openwrt/scripts/00-prepare_base.sh#L392

minkoonim, 2024-04-04:
> 不能，这个仓库你看到的所有文件它都是来自一个 caddy2 服务器上的备份。所有的目录结构仅用于方便区分文件用途，而不是直接针对 openwrt 工程目录。
> 
> 你能上传你自己的配置到 r4s_build_script/openwrt 位置，在任意一个 sh 文件中通过 curl 或 wget 下载到 openwrt 工程位置上，如
> 
> https://github.com/sbwml/r4s_build_script/blob/master/openwrt/scripts/00-prepare_base.sh#L392

谢谢!!

closed 2024-04-04 by minkoonim