sbwml/r4s_build_script #28: 老大，请教下首页CPU使用率无法显示的问题

lwb1978, 2024-05-09:
@sbwml 老大，我编译的immortalwrt 的主分支版本，功能都正常，就是首页的CPU使用率显示 ？ 号，不知道缺少了什么依赖没安装，查了半天没查出来，特跑到老大门下请教一下，望指教。

lwb1978, 2024-05-09:
[image]
忘了说了，是x86的版本

sbwml, 2024-05-09:
也许是缺少 rpcd 权限。我没玩过 imm 源码，具体是什么原因不清楚。

但是你可以找他们源码的 `getCPUUsage` 字符，正常会在某个  .json 文件存在 ubus 权限，如果没找到，那就是缺少了 rpcd 。

我用的  openwrt  但是cpu 占用率我也是从 imm 拿的 status ，不过是很多年前的事情了。

[image]

lwb1978, 2024-05-09:
破案了，是procps-ng-top的锅，而modules/luci-base/root/usr/share/rpcd/ucode/luci的代码片段如下，目前我要解决这个问题要不就是卸载procps-ng，要不就修改代码为：/bin/busybox top

        getCPUUsage: {
                call: function() {
                        const fd = popen('top -n1 | awk \'/^CPU/ {printf("%d%", 100 - $8)}\'');
                        if (fd) {
                                let cpuusage = fd.read('all');
                                if (!cpuusage)
                                        cpuusage = '?';
                                fd.close();

                                return { cpuusage: cpuusage };
                        } else {
                                return { cpuusage: error() };
                        }
                }
        },

sbwml, 2024-05-09:
[image]
这个确实

closed 2024-05-12 by lwb1978