sbwml/r4s_build_script #31: 安装 Clang/LLVM 工具链之后，下载的16-18特定的版本不会自动改默认

gitbruc, 2024-05-18:
需要把好多llvm工具都改成默认才行，比如llvm-ar-17改成默认的llvm-ar，怎么自动改，工具太多了

sbwml, 2024-05-18:
配置文件可以定义的    具体是什么配置     你看一下 commit 内容 

发自我的iPhone

gitbruc, 2024-05-18:
我是新手来的，老是报这种错make[1]: *** [package/Makefile:123: /home/bc/openwrt/staging_dir/target-x86_64_glibc/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bc/openwrt'
make: *** [/home/bc/openwrt/include/toplevel.mk:232: world] Error 2
我现在用的Clang/LLVM-17，编译的时候会自动选择clang-llvm不带版本号的，这种情况应该修改哪一个地方？

sbwml, 2024-05-18:
https://github.com/sbwml/r4s_build_script/blob/master/openwrt/generic/config-clang#L2

配置 `CONFIG_KERNEL_CC="clang-17"` ，如无意外它会自动使用 `-17` 后缀的 clang & llvm 工具链（在 kernel 的构建动作我是这样设计的，但是我自己没有真实测试过）。

或者你可以直接使用我GitHub 上提供的 clang 版本，这是专门为 openwrt clang 编译提供的：https://github.com/sbwml/redhat-llvm-project/releases

下载并解压，设置 bin 为系统变量即可使用。

gitbruc, 2024-05-18:
好我试一下！感谢大佬🙏🏻️

closed 2024-05-20 by gitbruc

sbwml, 2024-05-21:
最终你有没有测试成功，还是说失败了？

gitbruc, 2024-05-22:
用你的clang成功了大佬，之前一直失败是因为开了glibc（实验性）😂️，我编译了四五天关了glibc之后终于成功了。
但是我发现23.05的feeds/luci/modules/luci-mod-admin-full/这里面只有makefile，没有luasrc/view/admin_status/index.htm了，跟lede不一样，找了半天也没找到，LuCI的模板文件换位置了吗？

sbwml, 2024-05-23:
> 用你的clang成功了大佬，之前一直失败是因为开了glibc（实验性）😂️，我编译了四五天关了glibc之后终于成功了。
> 但是我发现23.05的feeds/luci/modules/luci-mod-admin-full/这里面只有makefile，没有luasrc/view/admin_status/index.htm了，跟lede不一样，找了半天也没找到，LuCI的模板文件换位置了吗？

lede是 luci1  （lua）   openwrt是luci2  它是js的

gitbruc, 2024-05-23:
那现在想修改luci模板要从哪里改呀？

gitbruc, 2024-05-23:
ok,我再研究研究，谢大佬🙏🏻️