sbwml/r4s_build_script #39: 请求添加mlx4_core和mlx5_core驱动程序

Q2297045667, 2024-07-08:
英伟达的ConnectX系列的网卡用的人应该挺多的

sbwml, 2024-07-08:
> 英伟达的ConnectX系列的网卡用的人应该挺多的 编译的时候把这个驱动编译进去

https://github.com/sbwml/r4s_build_script/commit/1980291feb8867a2b8a5bc7e4e36eb570e640919

X86 架构下一个构建会包含 `kmod-mlx4-core` & `kmod-mlx5-core` 驱动模块。在此之前的版本也可以通过软件中心搜索安装。

> 建议默认源使用如下源

这里的 openwrt 固件遵循 github.com/openwrt/openwrt  官方 23.05 libc 以及 system libraries，所以我只能官方软件源或来自openwrt 官方上游的镜像源。至于任何来历不明的软件源是不会直接应用到固件中的。

如果确实有这个需求，你可以自己修改设备的 opkg 安装源。

Q2297045667, 2024-07-08:
> > 英伟达的ConnectX系列的网卡用的人应该挺多的 编译的时候把这个驱动编译进去
> 
> [1980291](https://github.com/sbwml/r4s_build_script/commit/1980291feb8867a2b8a5bc7e4e36eb570e640919)
> 
> X86 架构下一个构建会包含 `kmod-mlx4-core` & `kmod-mlx5-core` 驱动模块。在此之前的版本也可以通过软件中心搜索安装。
> 
> > 建议默认源使用如下源
> 
> 这里的 openwrt 固件遵循 github.com/openwrt/openwrt 官方 23.05 libc 以及 system libraries，所以我只能使用官方软件源或来自 openwrt 官方上游的镜像源。至于任何来历不明的软件源是不会直接应用到固件中的。
> 
> 如果确实有这个需求，你可以自己修改设备的 opkg 安装源。

好的，万分感谢！

closed 2024-07-08 by Q2297045667