sbwml/r4s_build_script #41: 一个小问题 全锥型 NAT（NFT、BCM 双方案）

Q2297045667, 2024-07-13:
全锥型 NAT（NFT、BCM 双方案）中的BCM是由那个软件包实现的？
卸载了一部分用不到的软件吧之后发现博通的NAT没了
想重新安装一下
[image]

sbwml, 2024-07-13:
brcm-fullcone 由防火墙与内核实现，并非单个独立软件包。但 LuCI 支持上，brcm 没有做独立的功能展示，它必须与 nft-fullcone 功能同时存在（`kmod-nft-fullcone`）才会展示 全锥型 NAT 的 LuCI 菜单。

如果你在 “系统” - “软件包” - “已安装” 中卸载某些软件包导致的选项消失，只能重刷固件恢复功能。同时避免卸载或更新任何固件自带的 kmod、lib、firewall 相关内容。

Q2297045667, 2024-07-14:
> brcm-fullcone 由防火墙套件与内核实现，并非单个独立软件包。但 LuCI 支持上，brcm 没有做独立的功能展示，它必须与 nft-fullcone 功能同时存在（`kmod-nft-fullcone`）才会展示 全锥型 NAT 的 LuCI 菜单。
> 
> 如果你在 “系统” - “软件包” - “已安装” 中卸载某些软件包导致的选项消失，只能重刷固件恢复功能。同时避免卸载或更新任何固件自带的 kmod、lib、firewall 相关内容。

感谢回复，但是我注意到卸载软件包的时候并没有卸载软件包以外的包，因为我并没有勾选“自动移除未使用的依赖”

sbwml, 2024-07-14:
虽然我不知道你为什么要卸载内置的包，很显然我猜不到你到底卸载的什么和问题出现的原因。

不过没有关系，我可以给你推荐一个更优秀的固件，与本固件相比，它内置的软件包更少   https://github.com/QiuSimons/YAOF

sbwml, 2024-07-14:
> 虽然我不知道你为什么要卸载内置的包，很显然我猜不到你到底卸载的什么和问题出现的原因。
> 
> 不过没有关系，我可以给你推荐一个更优秀的固件，与本固件相比，它内置的软件包更少 https://github.com/QiuSimons/YAOF

同时它也支持 brcm fullcone 
[image]

Q2297045667, 2024-07-14:
> 同时它也支持 brcm fullcone [...]

只是闲着没事折腾而已，我自己Fork了一个，自己改一改 看看能不能解决这个问题，在编译的时候直接去掉不需要的软件包，不要抛弃我，嘿嘿

sbwml, 2024-07-14:
其实意义并不大，对于构建完全不熟悉的情况下，自己构建出来 和 你能直接下载到的固件，它们完全是两种东西来的

Q2297045667, 2024-07-14:
> 其实意义并不大，对于构建完全不熟悉的情况下，自己构建出来 和 你能直接下载到的固件，它们完全是两种东西来的

[image]
没事瞎折腾，我去看看构建结果

closed 2024-07-14 by Q2297045667