sbwml/r4s_build_script #54: R5S编译失败

Turing-China, 2024-10-26:
系统使用：Ubuntu 22.04.5 LTS
网络访问GITHUB正常
make[3] -C tools/sstrip compile
 make[3] -C tools/upx compile
 make[3] -C tools/zip compile
 make[3] -C tools/clang compile
 make[3] -C tools/mtools compile
    ERROR: tools/clang failed to build.
    ERROR: tools/upx failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/turing/openwrt/include/toplevel.mk:233：world] 错误 1

 Build error... 
 Build time: 0h,18m,11s

sbwml, 2024-10-26:
八成你网络问题，tools/clang 需要下载大约 600M 的文件，而 openwrt 下载文件是有超时限制的，规定时间内文件下载没有完成就失败。

这个东西支持GitHub action 在线编译，用不到本地浪费电费

发自我的iPhone

Turing-China, 2024-10-26:
> 八成你网络问题，tools/clang 需要下载大约 600M 的文件，而 openwrt 下载文件是有超时限制的，规定时间内文件下载没有完成就失败。 这个东西支持GitHub action 在线编译，用不到本地浪费电费 发自我的iPhone
> […](#)

tools/clang是我删除了openwrt目录重新构建的时候出现的，之前一直是tools/upx failed to build.
GitHub action 不是没办法自定义插件吗

sbwml, 2024-10-26:
> GitHub action 不是没办法自定义插件吗

你可以在 https://github.com/sbwml/r4s_build_script/blob/master/openwrt/scripts/10-custom.sh 自定义拉取你需要的插件源码

https://github.com/sbwml/r4s_build_script/blob/master/openwrt/23-config-common  这里可以声明你需要编译的插件

sbwml, 2024-10-26:
本地编译出错也就两点原因，1、网络  2、系统环境

Turing-China, 2024-10-27:
> > GitHub action 不是没办法自定义插件吗
> 
> 你可以在 https://github.com/sbwml/r4s_build_script/blob/master/openwrt/scripts/10-custom.sh 自定义拉取你需要的插件源码
> 
> https://github.com/sbwml/r4s_build_script/blob/master/openwrt/23-config-common 这里可以声明你需要编译的插件

非常感谢解答，我使用GitHub Action编译成功了,能否增加一个环境变量用于更改系统默认密码
10-custom.sh是否还是这种方式拉取源码
git clone https://github.com/coolsnowwolf/luci/tree/openwrt-23.05/applications/luci-app-syncthing package/luci-app-syncthing

Turing-China, 2024-10-27:
> > GitHub action 不是没办法自定义插件吗
> 
> 你可以在 https://github.com/sbwml/r4s_build_script/blob/master/openwrt/scripts/10-custom.sh 自定义拉取你需要的插件源码
> 
> https://github.com/sbwml/r4s_build_script/blob/master/openwrt/23-config-common 这里可以声明你需要编译的插件

git clone -b openwrt-23.05 https://github.com/immortalwrt/luci temp_luci && \
cp -r temp_luci/applications/luci-app-syncthing/* package/luci-app-syncthing/ && \
rm -rf temp_luci
麻烦问下这是路径没对吗？编译提示找不到package/luci-app-syncthing/

sbwml, 2024-10-27:
`10-custom.sh` 它会在 openwrt 源码目录下执行，这个目录里面包含文件夹列表：https://github.com/openwrt/openwrt/tree/openwrt-23.05

至于你加进来的东西它能不能被编译，需要自己解决

Turing-China, 2024-10-27:
> `10-custom.sh` 它会在 openwrt 源码目录下执行，这个目录里面包含文件夹列表：https://github.com/openwrt/openwrt/tree/openwrt-23.05
> 
> 至于你加进来的东西它能不能被编译，需要自己解决

我想知道是把插件源码拉取到那个目录呢  拉取到package/下好像不行

sbwml, 2024-10-27:
任意目录，通常建议 package/new/xxxx

举例：

编辑：`10-custom.sh`

```shell
#!/bin/bash

# 自定义脚本

git clone https://github.com/sbwml/luci-app-xunlei package/new/luci-app-xunlei
```

编辑：`23-config-common`

```
CONFIG_PACKAGE_luci-app-xunlei=y
```

sbwml, 2024-10-27:
ps：第三方插件编译的问题你应该去找插件的上游作者获取帮助。

closed 2024-10-27 by sbwml