sbwml/r4s_build_script #56: 在系统日志中dnsmasq产生大量重复信息

smdx, 2024-10-28:
大佬你好，在系统日志中dnsmasq产生大量重复信息，麻烦抽空看看日志是什么原因造成的，谢谢

编译设置：
default: 'BUILD_FAST=y ENABLE_BPF=y ENABLE_LTO=y ENABLE_LRNG=y ENABLE_MOLD=y ENABLE_DPDK=y ENABLE_UHTTPD=y KERNEL_CLANG_LTO=y USE_GCC14=y'

x86/64
OpenWrt 23.05.5 r24106-10cc5fcd00 / LuCI openwrt-23.05 branch git-24.264.56413-c7a3562
6.6.58

```
Mon Oct 28 10:26:39 2024 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Mon Oct 28 10:26:39 2024 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: started, version 2.90 cache disabled
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: DNS service limited to local subnets
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 10.0.0.2 -- 10.0.0.254, lease time 12h
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain hhyy100.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain sinochemitc.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain zhi-ming.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain cgjoy.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain mgenware.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain gobelieve.io
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain mxweiqi.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain lygtravel.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain houzislkdjfkldsdsd.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain yunexam.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain qlbg.net
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain maptalks.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain 17zhiliao.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain 8090cdn.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain seassoon.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain guizhounaishili.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain write-bug.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain jizhida.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain proya.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain qteng.net
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain inhdd.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain xw-scm.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain 4006055885.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain chinadougong.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain shuiliantong.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain newqjsteel.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain bjjinl.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain gd1580.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain dmeng.net
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain 000vw.com
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: more servers are defined but not logged
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 39 names
Mon Oct 28 10:26:43 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Mon Oct 28 10:27:02 2024 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Mon Oct 28 10:27:05 2024 daemon.info dnsmasq[1]: started, version 2.90 cache disabled
Mon Oct 28 10:27:05 2024 daemon.info dnsmasq[1]: DNS service limited to local subnets
Mon Oct 28 10:27:05 2024 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Mon Oct 28 10:27:05 2024 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Mon Oct 28 10:27:05 2024 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 10.0.0.2 -- 10.0.0.254, lease time 12h
Mon Oct 28 10:27:05 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335
Mon Oct 28 10:27:05 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Mon Oct 28 10:27:05 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Mon Oct 28 10:27:05 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Mon Oct 28 10:27:05 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Mon Oct 28 10:27:05 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
[... 95 lines cut ...]
Mon Oct 28 10:55:10 2024 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 10.0.0.2 -- 10.0.0.254, lease time 12h
Mon Oct 28 10:55:10 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335
Mon Oct 28 10:55:10 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain hhyy100.com
...
省略
...
...
省略
...
...
省略
...
Mon Oct 28 10:55:10 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain dmeng.net
Mon Oct 28 10:55:10 2024 daemon.info dnsmasq[1]: more servers are defined but not logged
Mon Oct 28 10:55:10 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Mon Oct 28 10:55:10 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Mon Oct 28 10:55:10 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Mon Oct 28 10:55:10 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Mon Oct 28 10:55:10 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Mon Oct 28 10:55:10 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Mon Oct 28 10:55:10 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Mon Oct 28 10:55:11 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Mon Oct 28 10:55:11 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 39 names
Mon Oct 28 10:55:11 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
```

sbwml, 2024-10-28:
这是正常的 dnsmasq 日志，前面部分的日志你应该是有开启 passwall 之类的科学插件，后面部分的日志也是正常的局域网设备与 dnsmasq 之间的 dhcp 交互（DHCPDISCOVER、DHCPOFFER、DHCPREQUEST、DHCPACK）

smdx, 2024-10-28:
感谢回复，请问这个相同的内容隔段时间就重复出现没问题吗？

sbwml, 2024-10-28:
> 感谢回复，请问这个相同的内容隔段时间就重复出现没问题吗？

没有问题，如果你不希望它显示这些信息，你可以尝试在 **网络** - **DHCP/DNS** - **高级设置** - **不记录日志**（禁止记录 DHCP 协议的日常操作。） 打开它。

smdx, 2024-10-28:
好的，感谢解答。

closed 2024-10-28 by smdx