sbwml/r4s_build_script #62: 软件包的更新

Turing-China, 2024-11-14:
大佬，麻烦问下想更新某个软件包的版本，默认的版本过低，又不想用开发版，这个脚本该如何操作呢

sbwml, 2024-11-14:
https://github.com/sbwml/r4s_build_script/blob/master/openwrt/scripts/10-custom.sh

这个脚本会在 openwrt 源码位置执行，你可以用你知道的任何方法更新 makefile 内容

Turing-China, 2024-11-15:
> https://github.com/sbwml/r4s_build_script/blob/master/openwrt/scripts/10-custom.sh
> 
> 这个脚本会在 openwrt 源码位置执行，你可以用你知道的任何方法更新 makefile 内容
我看build.sh 里面拉取的就是最新的软件包啊，但是为啥编译出来版本跟拉取的不一致，脚本没做任何改变，是我用了缓存的原因吗？
git clone https://$github/openwrt/packages master/packages --depth=1

sbwml, 2024-11-15:
你这个操作挺无语的

发自我的iPhone

Turing-China, 2024-11-15:
再10-custom.sh脚本里面修改  ./feeds/packages/net/某个软件/Makefile 里面的版本号和hash值 是不是也可以

Turing-China, 2024-11-17:
大佬，编译PHP8报错能看看吗？PHP8是官方源码自带的包 ERROR: package/feeds/packages/php8 failed to build.
没有输出错误的详细信息
[logs_30970658011.zip](https://github.com/user-attachments/files/17788434/logs_30970658011.zip)

sbwml, 2024-11-17:
你的提供的日志没有任何有效信息。

论算命的话，假设openwrt 官方 php 包没有问题，那么唯一可能会导致问题的就是来自非官方的 gcc 15 了，php8 代码不一定兼容 gcc15 编译器，你可以尝试将其更改为   GCC13 （openwrt 24.10 默认使用该版本）

https://github.com/sbwml/r4s_build_script/blob/master/.github/workflows/build-release.yml#L109

sbwml, 2024-11-17:
如果使用 GCC13 依然编译失败，那则考虑该包是否存在问题，你需要携带有效日志到 openwrt 上游仓库或openwrt 社区反馈。

因为我不会帮你去解决这种东西。因为我歧视在 op 固件使用  php 这种行为。

Turing-China, 2024-11-18:
> 如果使用 GCC13 依然编译失败，那则考虑该包是否存在问题，你需要携带有效日志到 openwrt 上游仓库或openwrt 社区反馈。
> 
> 因为我不会帮你去解决这种东西。因为我歧视在 op 固件使用 php 这种行为。

我改GCC13，刚一编译就报错了
[image]

sbwml, 2024-11-19:
修复 https://github.com/sbwml/r4s_build_script/commit/a84c446990030f68bb52382cd1a0796115c2ba06

closed 2024-11-21 by Turing-China