sbwml/r4s_build_script #65: 编译失败

ncwb, 2024-11-19:
make[2]: *** No rule to make target 'Not', needed by '404'.  Stop.
make[1]: *** [/builder/openwrt/include/toplevel.mk:[9](https://github.com/ncwb/openwrt-x86-64/actions/runs/11905928866/job/33177183139#step:10:10)7: prepare-tmpinfo] Error 2
make: *** [/builder/openwrt/include/toplevel.mk:246: world] Error 2

重新fork后 用自带的yml文件编译 问题仍然复现

sbwml, 2024-11-19:
GitHub 连接出现问题，等晚上的，改成本地 http 服务。

发自我的iPhone

sbwml, 2024-11-19:
修复 https://github.com/sbwml/r4s_build_script/commit/a84c446990030f68bb52382cd1a0796115c2ba06

ncwb, 2024-11-19:
已成功编译验证 感谢！

ncwb, 2024-11-20:
昨晚添加无线网卡驱动后出现package编译失败
    ERROR: package/kernel/rtl8812au-ct failed to build.

sbwml, 2024-11-20:
删掉这个源码或者 is not set 即可，rtw88 的 8812au  已经完全适应于openwrt   而不仅限客户端模式

发自我的iPhone

Turing-China, 2024-11-21:
> 删掉这个源码或者 is not set 即可，rtw88 的 8812au 已经完全适应于openwrt 而不仅限客户端模式 发自我的iPhone
> […](#)

 is not set 还是会提示这个
    ERROR: package/kernel/rtl8812au-ct failed to build.

closed 2024-11-23 by sbwml