sbwml/r4s_build_script #77: 关于英特尔网卡驱动

shuangqiao999, 2024-12-09:
你好 ixgbe驱动是最新版吗？如果要更新驱动要修改哪里？

sbwml, 2024-12-09:
`kmod-ixgbe` 属于 in tree 驱动，它的版本来自 Linux 6.12。

https://github.com/sbwml/r4s_build_script/blob/7fc2f5521554dc4be08fccb9dfee67a1bc49c9c5/openwrt/patch/openwrt-6.x/modules/netdevices.mk#L1078-L1094

>> 你好 ixgbe驱动是最新版吗？

但是如果你问它是不是最新版，那么它一定不是。因为驱动每时每刻都在 linux-next 中更新。但是这里的 `kmod-ixgbe` 驱动源码比你在任何渠道能下载到的 OpenWrt 都要新。

>> 如果要更新驱动要修改哪里？

两个方法：

1、backport 上游更新 https://git.kernel.org/pub/scm/linux/kernel/git/tnguy/next-queue.git
2、自行维护一个内核版本（如 Linux 6.13、6.14 等，始终跟随内核主线版本，那么你会的到最接近最新版本的驱动源码）

shuangqiao999, 2024-12-09:
好的 谢谢老大科普

sbwml, 2024-12-11:
> 大佬何时考虑同步linux6.13

Linux 6.12 是 LTS 版本，会用一年，即下一个 LTS 版本出来时。

至于其它的稳定版这里是不会使用的。或许你可以去 lede 那些仓库看看，可能有你想要的内核版本。

发自我的iPhone

closed 2024-12-16 by sbwml