sbwml/r4s_build_script #91: 检测到可能的DNS重新绑定攻击

shuangqiao999, 2025-01-12:
老大 你好 
这个是什么情况  这是日志
daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:16:51 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:17:09 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:17:14 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:17:17 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:17:43 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:18:40 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:18:48 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:18:52 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:19:04 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:19:08 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:19:11 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:19:15 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:19:15 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:19:50 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:19:54 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:19:58 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:20:00 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:20:05 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:20:08 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:20:17 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:20:19 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:20:23 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net
Sun Jan 12 10:20:33 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: whoami.akamai.net

shuangqiao999, 2025-01-12:
这是有人在攻击路由器吗？

sbwml, 2025-01-12:
关闭 **重绑定保护**

shuangqiao999, 2025-01-12:
好的 谢谢🙏🙏🙏

closed 2025-01-12 by sbwml