sbwml/r4s_build_script #103: KMS插件好像有点问题

TuringTrail, 2025-04-02:
启用保存应用后状态显示VLMCSD 未运行
重启路由器状态显示正常，但是实际服务未启动1688端口不通
[image]

sbwml, 2025-04-02:
> TuringTrail
>  created an issue
> (sbwml/r4s_build_script#103)
> 启用保存应用后状态显示VLMCSD 未运行
> 重启路由器状态显示正常，但是实际服务未启动1688端口不通
> image.png (view on web)

命令

/usr/bin/vlmcsd -D -i /etc/vlmcsd/vlmcsd.ini -L [::]:1688

发自我的iPhone

TuringTrail, 2025-04-02:
> 命令
> 
> /usr/bin/vlmcsd -D -i /etc/vlmcsd/vlmcsd.ini -L [::]:1688
> 
> 发自我的iPhone
> […](#)
> 
> 
>  
> 
>  TuringTrail
>   created an issue
>  ([sbwml/r4s_build_script#103](https://github.com/sbwml/r4s_build_script/issues/103))
>  启用保存应用后状态显示VLMCSD 未运行
>  重启路由器状态显示正常，但是实际服务未启动1688端口不通
>  image.png (view on web)

[image]
大佬 还是不行呀

sbwml, 2025-04-02:
那就是这个东西有问题了  

[image]

sbwml, 2025-04-02:
修复 https://github.com/sbwml/openwrt_pkgs/commit/f08e1e71e210a07e07cea33465dce3cea8624a8e

TuringTrail, 2025-04-02:
> 修复 [sbwml/openwrt_pkgs@f08e1e7](https://github.com/sbwml/openwrt_pkgs/commit/f08e1e71e210a07e07cea33465dce3cea8624a8e)

感谢大佬

closed 2025-04-02 by TuringTrail