sbwml/r4s_build_script #104: 编译失败

VFool, 2025-04-15:
config文件不修改的情况下编译失败
仓库已经填好了环境变量
[编译失败的地址](https://github.com/VFool/r4s_build_script/actions/runs/14457978934/job/40545079049)
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-phy-bcm-ns-usb2', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-phy-bcm-ns-usb3', which does not exist
make[2]: Entering directory '/builder/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/builder/openwrt/scripts/config'
make[1]: Entering directory '/builder/openwrt'
make[2]: Entering directory '/builder/openwrt'
+ mkdir -p /builder/openwrt/staging_dir/target-mips_24kc_musl
+ cd /builder/openwrt/staging_dir/target-mips_24kc_musl
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /builder/openwrt/build_dir/target-mips_24kc_musl/stamp
touch /builder/openwrt/staging_dir/target-mips_24kc_musl/.prepared
make[3]: Entering directory '/builder/openwrt/target/linux'
make[4]: Entering directory '/builder/openwrt/target/linux/ath7[9](https://github.com/VFool/r4s_build_script/actions/runs/14457978934/job/40545079049#step:12:10)'
rm -f /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/localversion
/builder/openwrt/scripts/kconfig.pl  + + /builder/openwrt/target/linux/generic/config-6.6 /builder/openwrt/target/linux/ath79/config-6.6 /builder/openwrt/target/linux/ath79/generic/config-default > /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /builder/openwrt/.config >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.target
/builder/openwrt/scripts/package-metadata.pl kconfig /builder/openwrt/tmp/.packageinfo /builder/openwrt/.config 6.6 > /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.override
/builder/openwrt/scripts/kconfig.pl 'm+' '+' /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.target /dev/null /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.override > /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
mv /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.old
grep -v INITRAMFS /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.old > /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo "# CONFIG_INITRAMFS_PRESERVE_MTIME is not set" >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo '# CONFIG_FSCACHE is not set' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
mv /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.old
grep -v CONFIG_LTO /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.old > /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LTO=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LTO_CLANG=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LTO_CLANG_THIN=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_HAS_LTO_CLANG=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_RANDSTRUCT_NONE=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo '# CONFIG_CFI_CLANG is not set' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo '# CONFIG_LTO_CLANG_FULL is not set' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo '# CONFIG_LTO_NONE is not set' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo '# CONFIG_RANDSTRUCT_FULL is not set' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo '# CONFIG_RELR is not set' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_DEBUG_INFO_BTF_MODULES=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_PROBE_EVENTS_BTF_ARGS=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_AUTO_SELECTED=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_COLLECTION_SIZE_[10](https://github.com/VFool/r4s_build_script/actions/runs/14457978934/job/40545079049#step:12:11)24=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_CONTINUOUS_COMPRESSION_ENABLED=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_CPU=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_CPU_ENTROPY_RATE=8' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_DEV_IF=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_DFLT_DRNG_CHACHA20=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_IRQ=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_IRQ_DFLT_TIMER_ES=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_IRQ_ENTROPY_RATE=256' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_JENT=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_JENT_ENTROPY_BLOCKS=[12](https://github.com/VFool/r4s_build_script/actions/runs/14457978934/job/40545079049#step:12:13)8' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_JENT_ENTROPY_BLOCKS_NO_128=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_JENT_ENTROPY_RATE=[16](https://github.com/VFool/r4s_build_script/actions/runs/14457978934/job/40545079049#step:12:17)' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_KCAPI_IF=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_RUNTIME_ES_CONFIG=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_SCHED=y' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
echo 'CONFIG_LRNG_SCHED_ENTROPY_RATE=4294967295' >> /builder/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.6.73/.config.set
[... cut ...]
time: target/linux/compile#0.85#1.37#2.[18](https://github.com/VFool/r4s_build_script/actions/runs/14457978934/job/40545079049#step:12:19)
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:32: target/linux/compile] Error 1
make[2]: Leaving directory '/builder/openwrt'
make[1]: *** [target/Makefile:25: /builder/openwrt/staging_dir/target-mips_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/builder/openwrt'
make: *** [/builder/openwrt/include/toplevel.mk:[23](https://github.com/VFool/r4s_build_script/actions/runs/14457978934/job/40545079049#step:12:24)3: world] Error 2

sbwml, 2025-04-15:
> VFool
>  created an issue

不再提供构建。（ https://github.com/VFool/r4s_build_script?tab=readme-ov-file#由于本源码具备后门被证实-92良心发现后防止毒害社会不再允许任何人与狗直接构建 ）

换一个，个人推荐404佬的    https://github.com/QiuSimons/YAOF

发自我的iPhone

VFool, 2025-04-15:
好的，谢谢大佬提醒